ipa320/schunk_modular_robotics #142: ROS Jade Release

jpetereit, 2015-07-21:
Desperately waiting for it :)
Is there any blocking issue or anything I can help you with?

ipa-fxm, 2015-07-21:
@jpetereit 

I made some Jade tests a while ago (in May). You can find `jade_dev` branches in my forks!

I was able to compile all cob_* repositories but I had to create a lot of overlays for packages that haven't been released into jade at that time - mainly driver packages for cameras and other sensors - but also there where some issues related to `gazebo_ros_pkgs` and `ros_control`/`ros_controllers`.

Anyway, we are mainly developing cob_* packages for Indigo at the moment. Thus, `jade_dev` currently has only secondary priority.

I can have another look at my `jade_dev` branches and update them to latest commits to `indigo_dev`...but I can't promise anything.

Feel free to use my `jade_dev` as a basis and report your experience here!
Thanks for you interest!

ipa-fmw, 2015-07-22:
dear @jpetereit, we've decided to officially only support LTS versions of ROS which is currently Indigo. As Jade is no LTS we will NOT release this Package for Jade but only do migration changes in jade_dev to prepare for ROS-K release (which is a LTS version again). 

Are you able to work with the compilation from source for Jade? or does this block you totally to use this package? 

We're of course happy to integrate fixes in form of pull requests for jade_dev.

jpetereit, 2015-07-23:
I was partially able to compile the cob_* packages from source, but as you said: There are still some other dependencies missing from the Jade release and I wasn't in the mood to create all those needed source overlays yet. :)

Maybe, we can use this issue to maintain a list of the missing dependencies for reference? I'll start with the packages that I have tried to compile:

#### cob_common
*build successful*

#### cob_control
Missing dependencies:
- [ ] controller_interface
- [ ] controller_manager_msgs
- [ ] hardware_interface

#### cob_extern
*build successful*

#### schunk_modular_robotics
*build successful*

ipa-fmw, 2015-07-23:
thanks for the feedback. 
For reference: https://github.com/ros-controls/ros_control/issues/201 discusses the ros_control Jade release which is a prerequisite for schunk_modular_robotics

ipa-fxm, 2015-08-25:
Jade Releases will be investigated anew after the latest Indigo Releases (see [here](https://github.com/ipa320/care-o-bot/issues/7#issuecomment-134714774)) have gone through....

ipa-fxm, 2015-09-04:
Latest Indigo Release is available in shadow-fixed...will be available from debs with next sync.
We'll try the Jade Release once https://github.com/ros-controls/ros_control/issues/201 is resolved.

closed 2016-04-02 by ipa-fxm