ipa320/schunk_modular_robotics #143: [question] Overheating issues with the Schunk SDH hand?

iamwolf, 2015-07-29:
Hey, 
Thank you very much for a fantastic driver package! 
We are using the Schunk SDH hand with your ROS driver. Since we are suffering from frequent and severe overheating (hand stops responding to input commands without throwing an exception) after about 30mins of usage, we have expanded your SDH driver with service calls to engage and disengage the motors (calling respective Schunk SDH API calls). Since overheating happens very frequently for us (just the hand being powered, not in use - and sometimes even when motors are disengaged), we are wondering whether you have seen this issue with Schunk SDH hands back when you used them on Care-O-Bots or what you have done to mitigate this issue - are there any secret tricks? ;)

Also, did you see any issues when initialising the hand in the beginning? When calling the /init service, initialising fails ~one out of five times.

Thank you very much in advance :)

ipa-fmw, 2015-08-03:
we've noticed that the hand always gets quite hot, but never noticed any interrupt due to high temperature.

Did you encounter any temperature difference with engaged and disengaged motors?

iamwolf, 2015-08-03:
Thanks for your response. We did encounter differences in how long we can operate the hand when disengaging motors when not in use (from 20-30mins to 1.5-2h), and it's subjectively less hot. 

Our previous driver implementation (before switching to this ROS driver) actually engaged motors before every motion, and then disengaged them afterwards. We've seen random crashes every now and then, which we also do with this driver since we've added the engaging and disengaging. Hence, we presume it's coming from the underlying SDH library.

closed 2015-08-03 by iamwolf

ipa-fxm, 2015-08-03:
@iamwolf you said you added the engaging/disengaging strategy to the driver...
Is there anything worth playing back to this reopository?
Keep us in the loop, in case you discover something during your further debugging...Thanks.

iamwolf, 2015-08-03:
We added quite a bit more as well (reinit, basic tactile grasping and opening without pre-shaping given that the SDH library has bold warnings around their tactile grasping implementation), so need for cleaning up before making a PR back (likely towards/not before the beginning of September). Will keep you posted on further debugging.

Are you still actively using the SDH hands now that the CareOBot 4 is out?

ipa-fmw, 2015-08-03:
we're not using them on cob4 but they are still in use on various cob3 robots.