ipa320/schunk_modular_robotics #154: [lwa-4d] Rotation mismatch

athackst, 2016-01-19:
Hi, 
I'm seeing a different axis of rotation on the prl+ joints from what the URDF model indicates.  In particular the axis of rotation seems to be pointing toward the base of the robot (positive delta = clockwise rotation) as opposed to what the URDF indicates  -- rotation pointing toward the end effector (positive delta = counter clockwise rotation).  I have checked to make sure my arm is using the default settings.  Are other people having this issue?  Should the model be updated to match the arm or is it better if I flip the direction of the motor in firmware?  Schunk seemed to indicated changing direction in firmware would not be ideal.

ipa-fxm, 2016-01-19:
I don't know in what state the LWA4D robots are shipped by the manufacturer, at IPA however, we tried to follow the (Denavit-Hartenberg) convention, letting the axis of rotation of a joint point towards the next joint (towards the end-effector!). Thus, you might want to check the PRL modules with the Schunk configuration software what is set for the direction of rotation and adjust accordingly...
Hope this helpfs!

athackst, 2016-01-22:
The frames can point the same direction, but I think the <axis> tag should be negated <axis xyz="0 0 -1"/>

Do you have any LWA4D robots you could check?

ipa-fxm, 2016-01-22:
I think this has been fixed in indigo_dev already...but it is likely that it hasn't been released yet.
I actually tested and modified this when working with our HW. Are you using debs or the source?
Can't see any -1 in the source for lwa

athackst, 2016-01-22:
I'm using the source from this repo.  I think it should have -1's, but it doesn't.

ipa-fxm, 2016-01-22:
why would you want -1? DH say to point positive z-axis towards next joint, isn't it?

athackst, 2016-01-22:
Because that's the direction of the default orientation of the motors in the arm.  So it's whether or not the URDF should match the hardware or the convention.  Do you guys really adjust all your hardware?

ipa-fxm, 2016-01-22:
Maybe we did it and then did not know about the default state anymore. At least I don't. Can't tell atm how it should be.. Our HW is used by many people.

ipa-fxm, 2016-01-22:
It just looked consistent this way

athackst, 2016-01-22:
If you look at the firmware it would report "invert motor: yes" and "invert position: yes" if you did.  Otherwise the default is false for both.

athackst, 2016-01-25:
I think it is probably best to leave it up to the user.  I added a parameter to the macro which will let you decide what is desired.  The default value follows the DH convention like before so no changes are forced upon others.

PR #155 (2016-02-05): added 'invert' parameter to lwa4d xacro description
default value 1 matches DH params, -1 matches default hardware configuration.
fixes #154
closed 2016-02-05 by ipa-fxm