ipa320/schunk_modular_robotics #172: [schunk_description] sdh explodes in gazebo 2.2.3

guihomework, 2016-09-16:
I instantiated a sdh model at the top of my lwr in gazebo on indigo, and got fingers exploding, usually coming from inter-penetrating collision objects.

[image]

I noticed that the palm and knuckle collision objects somehow inter-penetrate and are also quite detailed for collision objects.

Did you notice a similar behaviour ?

I will provide a PR with simplified collision models that do not inter-penetrate and hence with a model that does not explode in my gazebo simulation.

guihomework, 2016-12-21:
Retested, and noticed a few things :
* happens only on gazebo 2.2.3, not on gazebo 7 (kinetic)
* happens if one adds gazebo_ros_control plugin (so might also be fixed by the kinetic version of this plugin not only gazebo version)
* can be fixed with palm_collision change only

ipa-fxm, 2017-01-11:
Merged...Thanks!

closed 2017-01-11 by ipa-fxm