ipa320/schunk_modular_robotics #174: [schunk_description] sdh has wrong bad inertial

guihomework, 2016-09-20:
Currently sdh has default inertial terms, which create some huge inertia and then bad arm behavior when controlled in torque mode gazebo.

Compared to a shadow hand inertial, one can see with gazebo inertia/mass display that it does not match the model.

[image]

How accurate should be a fix I am willing to provide ? I don't know were to find individual masses of each sdh element, but could give estimates and use correctly parametrized box inertial macro calls instead of default.

What do you think ?

guihomework, 2016-09-21:
I used meshlab with the original stl, and following the [gazebo tutorial](http://gazebosim.org/tutorials?tut=inertia&cat=build_robot) to compute inertia and mass parameters from there. It is not so well explained on the scaling and material density level, but I think I have something acceptable now considering an all aluminum hand except a fake density for the palm to match the overall 1.95 kg.

If you like this https://github.com/ubi-agni/schunk_modular_robotics/commit/205ca05de8f6a2677a3605c6d1944b7b7bed61e6

and the result
[image]

I could do a PR

ipa-fxm, 2016-12-15:
@guihomework 
Please propose the PR

guihomework, 2016-12-20:
@ipa-fxm Sorry for the delay, very busy recording data.
Will try to have a look at the exploding issue before the end of the week hopefully.

ipa-fxm, 2017-01-11:
I merged #180...Thanks!

closed 2017-01-11 by ipa-fxm