ipa320/schunk_modular_robotics #175: unknown macro name: xacro:default_inertial

zainmehdi, 2016-10-04:
Hi,
   I am trying to make urdf file out of lwa.urdf.xacro ... When I am running the xacro.py command with following script I am getting the error. Can you please clarify whats wrong with it .

""" This is the code I am using for generating URDF file out of lwa.urdf.xacro"
<?xml version="1.0" ?>
<robot name="my_robot" xmlns:xacro="http://ros.org/wiki/xacro">

  
  <xacro:include filename="$(find schunk_description)/urdf/lwa/lwa.urdf.xacro"/>

  
  <link name="world" />

  
  <xacro:schunk_lwa parent="world" name="my_lwa">
    <origin rpy="0 0 0" xyz="0 0 0" />
  </xacro:schunk_lwa>
</robot>

ipa-nhg, 2016-10-05:
you need a file like https://github.com/ipa320/schunk_robots/blob/indigo_dev/schunk_lwa4p/urdf/robot.urdf.xacro where you include the lwa.urdf.xacro, the common.xacro and the materials.urdf.xacro , and define a root link ("world") as parent link of the lwa arm.

zainmehdi, 2016-10-06:
Thanks alot for the hint. And one more thing that the tag
 <xacro:default_inertial/>
isnt defined and creates and error as unknown xacro. I compiled my model after removing it from the code. Do I have to define it myself using assumed Inertia values or their are some predefined values for the lwa robots..
Again thanks alot for you precious time .

zainmehdi, 2016-10-06:
thnx alot :)

closed 2016-10-06 by zainmehdi