ipa320/schunk_modular_robotics #187: Support for Schunk Motion Protocol

manuelmuehlig, 2017-05-22:
Hi,

it seems that the new Schunk PTUs such as the PW 70 v6 only support the Schunk Motion Protocol and not the Amtec protocol anymore. Is there a plan about supporting the new protocol in this package?

I have contacted Schunk about if there is maybe an updated "ibm5api", implementing the same interface, but with updated communication messages. I got no reply yet, but I fear there is nothing yet. I consider writing a wrapper myself, but wanted to ask here first.

Best regards,
Manuel

ipa-fxm, 2017-05-23:
we at IPA currently do not use any of these Schunk devices...
but in case a PR is provided, we are happy to host it in this repository.

manuelmuehlig, 2017-05-24:
It will take a bit of time to submit a high-quality driver covering all functionality. I got a reply from Schunk and there is no libm5api-compatible implementation of the SMP. This means I have to write it from scratch and for my use case, I will start with only a minimal version only supporting basic control of the PW 70.

ipa-fxm, 2017-07-03:
@manuelmuehlig 
The PR https://github.com/ipa320/schunk_robots/pull/77 provided by @destogl has been merged now!

Feel free to build on top and extend the package to your needs...

hanjuns, 2017-10-11:
Hi,

I am using PW 70 V6 and trying to run this in ROS.

I installed the package for PW70 and ran ```robot.launch``` and ```dashboard.launch```.
But when I try to initialize uinsg the 'init' button on the dashboard, it gives me an error that says "Could not reset module 13 during init."

ROS version is indigo and we have IXXAT CAN module.

Can you help me how to solve this problem?

Thanks,
Hanjun

manuelmuehlig, 2017-10-12:
Hi Hanjun,

The v6 uses the Schunk Motion Protocol instead of the older Amtec protocol.
The software in this repository will not work with it. Unfortunately I did
not have the time to write a new driver but locally I am using a quite
hacky Python script directly sending CAN messages to the netcan device.

Greetings,
Manuel