ipa320/schunk_modular_robotics #190: Missing libSDHLibrary-CPP.so in ros-indigo-schunk-sdh

christian-rauch, 2017-07-15:
The package `ros-indigo-schunk-sdh` installs the executables:
```
/opt/ros/indigo/lib/schunk_sdh/dsa_only
/opt/ros/indigo/lib/schunk_sdh/schunk_sdh
/opt/ros/indigo/lib/schunk_sdh/sdh_only
```
The library `libSDHLibrary-CPP.so` that is used by these executables is however not installed:
```
$ ldd /opt/ros/indigo/lib/schunk_sdh/schunk_sdh
	../common/lib/libSDHLibrary-CPP.so => not found
```
and hence they cannot be used.

ipa-fxm, 2017-07-16:
is fixed in https://github.com/ipa320/schunk_modular_robotics/commit/523e3271cd660ed63bfd31cff143e3de4c12f20e...just hasnt been released since then...

closed 2017-07-16 by ipa-fxm

ipa-fxm, 2017-07-21:
a new release is in the pipeline for both [indigo](http://repositories.ros.org/status_page/ros_indigo_default.html?q=schunk_modular_robotics] and [kinetic](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=schunk_modular_robotics)