ipa320/schunk_modular_robotics #206: Melodic release

jschleicher, 2018-07-11:
Could you please consider releasing (at least) the `schunk_description` package to `melodic`?

fmessmer, 2018-07-23:
maybe we can release to melodic but ignore failing packages:
https://answers.ros.org/question/66545/to-ignore-some-pkgs-upon-release-by-bloom/

I will look into this next WE

agutenkunst, 2018-08-07:
Any updates on this?

ipa-nhg, 2018-08-16:
Done!
 https://github.com/ipa320/schunk_modular_robotics-release#schunk_modular_robotics-melodic---0612-0
Released for melodic ignoring some packages (thanks @fmessmer for the suggestion) . PR to rosdistro is already open https://github.com/ros/rosdistro/pull/18854

fmessmer, 2018-08-16:
as you bumped the version....please also release `0.6.12` into `indigo` and `kinetic`

ipa-nhg, 2018-08-16:
@fmessmer done!

closed 2018-09-13 by jschleicher