ipa320/schunk_modular_robotics #209: Gazebo dependency of schunk_description?

wxmerkt, 2018-11-16:
Hi,
Is the exec_depend on Gazebo for schunk_description required? It's a pretty heavy dependency when installing a robot model - I assume people attempting to simulate the robot and including the URDF for simulation will have the Gazebo dependency on that package. I.e., can we remove the dependency from the description package?

Thanks,
Wolfgang

jschleicher, 2019-07-12:
Would you mind dropping the gazebo dependency? The launch files that require gazebo are located in schunk_robots (e.g. [lwa4p](https://github.com/ipa320/schunk_robots/blob/indigo_dev/schunk_lwa4p/package.xml) with gazebo_ros dependency), so I'd propose to keep the footprint of `schunk_description` as small as possible.

@ipa-nhg @christian-rauch

christian-rauch, 2019-07-12:
I agree that the `exec_depend` on `gazebo_ros` is not required to use the URDFs. Unfortunately, there are no "optional" dependencies in catkin packages.
@jschleicher Can you create a PR that removes these dependencies?

PR #213 (2019-07-15): drop gazebo_ros dependency
Fixes #209
commit 0c5e2f99 (2019-07-15): drop gazebo_ros dependency

Fixes #209
closed 2019-07-15 by christian-rauch