ipa320/schunk_modular_robotics #211: use source version of SDHLibrary-CPP

christian-rauch, 2019-04-22:
We encountered communication issues with the SDH and DSA controller that was somehow related to the binary distributed driver in this node (`libSDHLibrary-CPP.so`). We solved the issue by building the driver (`SDHLibrary-CPP`) directly from source via a catkinised pure CMake package.

Are you willing to replace the binary library by the source version of the SDH driver? If so, could you create an empty repository at https://github.com/ipa320/SDHLibrary-CPP to hold the SDH driver? I will then open PR for the driver and the node.

fmessmer, 2019-04-25:
@christian-rauch 
thanks for all your contributions to this repo...

unfortunately, @ipa320 is actually no longer using SDH grippers - as far as I can tall.
for my part, I don't work for ipa320 anymore - but https://github.com/mojin-robotics/, a spin-off for the Care-O-bot 4 - which is why am still active on ipa320 repos

knowing what it feels like to not get PRs merged and stuff, I could think of the following solution:
as you seem to work with this gripper quite extensively, why not give you Contributor rights to this repo?
would that help you/be an option for you?

I will have to check with the ipa320 maintainers, though
@ipa-jba @ipa-fez @ipa-nhg

christian-rauch, 2019-04-25:
@fmessmer This would certainly help to get the fixes for the issues that we encounter upstream.
Is anybody from @ipa320 or @mojin-robotics still using the SDH?

If you agree to grant access rights, could I also gain the necessary rights to the release repository?
Also, if the source version of the SDHLibrary-CPP should be used, it would be useful to have it in its own repo. I would then just add it as git submodule to schunk_modular_robotics to be part of the release.

ipa-fez, 2019-04-29:
Hi, I will discuss it with @ipa-jba, we will let you know once we've decided.

ipa-fez, 2019-05-02:
@christian-rauch I've added you as collaborator to this repo and the release repo.

christian-rauch, 2019-05-02:
Thanks @ipa-fez 
Would it be possible to create a repo at https://github.com/ipa320/SDHLibrary-CPP (or similar) to store the source version of the library as CMake package?

christian-rauch, 2019-05-02:
@ipa-fez I am unable to merge pull requests via the GitHub web interface. What else would be required to merge my own PR (instead of pushing directly to a branch)?

ipa-fez, 2019-05-02:
You need an approved review before you can merge, but you should be able to review your own code.

For the library, it is probably easiest from a management standpoint if you just create the repo yourself.

christian-rauch, 2019-05-03:
@ipa-fez I am sorry to bother you with this further, but I am unable to merge changes:
- One cannot approve its own PR:
[image]
- I am also unable to merge manually outside of GitHub and push to the `indigo_dev` branch:
`! [remote rejected] indigo_dev -> indigo_dev (protected branch hook declined)`

I think as a collaborator, I am only able to review other's PRs and push to non-protected branches.

ipa-fez, 2019-05-03:
I've lifted the review restrictionm so you should be able to merge now.

closed 2019-06-11 by christian-rauch