ipa320/schunk_modular_robotics #220: Exploding SDH in with both ROS Kinetic and Melodic

padmaja-kulkarni, 2020-02-12:
Hello!

I have seen the previous issue about exploding SDH and I also saw that it is fixed. Now, I am using SDH with the current repository and SDH looks as in the image. 
Is this a known problem? Is there any solution?

[image]

christian-rauch, 2020-02-12:
I've not used the SDH2 model in Gazebo yet.
Which Gazebo and ROS version are you using? Is there a launch file with which I can reproduce this?

I had similar issues with other models in Gazebo. This looks more like that links are initialised at identity transform and not like a physical effect from gravity or other forces.

padmaja-kulkarni, 2020-02-13:
[Here](https://drive.google.com/open?id=11No5cm0zE7icj_UxkOmRQxk9BzEvWMLR) is the package I am using. README has the launch instructions.

Thanks!

christian-rauch, 2020-02-13:
I can reproduce the issue when I just "Play" the scene.

But I get a different result when stepping through the scene:
[image]

This might indeed be related to the "exploding" issue that was encountered in the past. Can you check if there are interpenetrating meshes?

padmaja-kulkarni, 2020-02-13:
Hi Christian! 

Thank you! I am tried using a small Geometry box instead of collision mesh files, the results stayed the same. I am not sure how to check for interpenetrating meshes!

christian-rauch, 2020-02-13:
As per the original issue https://github.com/ipa320/schunk_modular_robotics/issues/172 and PR https://github.com/ipa320/schunk_modular_robotics/pull/173 you could try to further reduce the size of the collision meshes. The original issue also mentions that this only happens with `gazebo_ros_control`.

If this is not caused by interpenetrating collision meshes, then I also don't know how to further approach this issue.

MatthiasNieuwenhuisen, 2020-04-19:
This problem is most probably caused by too small values for the inertia of the components which let the simulation become unstable. Increasing the inertia values solves this particular problem. For now I have only tested it with values several orders of magnitude too large, which might have other effects.