ipa320/schunk_modular_robotics #222: Noetic release

jschleicher, 2020-09-23:
Would you mind releasing especially schunk_description into noetic, such that the meshes and URDF files are available?

Is there anything blocking a noetic release that requires help?

christian-rauch, 2020-09-23:
If possible, I would like to release the entire repo at once. For this, `cob_common` has to be released to `noetic` as well: https://github.com/ipa320/cob_common/issues/285.

The source code ill build just fine on Ubuntu 20.04 (apart from warnings in `sdhlibrary_cpp`), so there are no source changes required.

christian-rauch, 2020-09-29:
https://github.com/ros/rosdistro/pull/26746

christian-rauch, 2020-09-29:
@jschleicher The package should now be in `ros-testing`. Can you check if this works for you and close this issue if applicable?

fmessmer, 2020-09-29:
@christian-rauch 
did you update the CMakeLists according to http://wiki.ros.org/noetic/Migration?
i.e. bump cmake version, python3 compatibility (if applicable), setuptools (if applicable), etc...

christian-rauch, 2020-09-29:
@fmessmer No, it just compiled fine without CMake warnings. I got some gcc warnings that might become errors on newer versions. Also, [`simulated_tactile_sensors.py`](https://github.com/ipa320/schunk_modular_robotics/blob/kinetic_dev/schunk_simulated_tactile_sensors/ros/src/simulated_tactile_sensors.py) , the only python script in the repo, seems to run fine.

jschleicher, 2020-10-01:
Yep, works great - tested at least in simulation / fake_execution. Thank you for your quick response!

closed 2020-10-01 by jschleicher